KurtE/Arduino_Phoenix_Parts #2: Support for PCA9685 i2C 16ch servo controller (Ebay and adafruit)

mano1979, 2017-05-16:
Is there support for PCA9685 i2C 16ch servo controller in your version of the phoenix code? Or maybe in the near future?

Will this also run on other arduino's than the teensy?

KurtE, 2017-05-18:
Sorry these days I am doing very little with the Phoenix code base, although keep meaning to get back to it. 

And for the most part I have no interest in working with RC servos... Using mostly Robotis servos these days. 

However if anyone wishes to implement it for this Servo Driver and issue a Pull request it would be great.

As for Teensy only?  No, should work with most Arduinos.  The code originally was done for the Lynxmotion board Botboarduino (Atmega 328 like UNO), have run on Megas as well as on the Trossen Arbotix boards,... At one point it was running on Chipkit M32 boards (not sure if that works any more).

Good Luck

KurtE, 2017-05-18:
If it were me trying to do this, I would probably start off with looking at the Adafruit library: 
https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library
They have an example program that controls some servos.

The example has a function that in theory can convert seconds (float) into the right value to output of I2C.
If it were me, I would probably change that function to take an int or uint16_t of microseconds and convert it into the appropriate value...

Then I would look how the SSC-32 driver converts the desired angle into pulse width in microseconds and then combine the two parts to figure out the right values to send out...  I would also experiment with then seeing how close I am... Send out a request for 45 degrees how close are we...

mano1979, 2017-05-18:
I see. I will look into it. Thanks for your reply.

keshav-s7, 2020-07-25:
> If it were me trying to do this, I would probably start off with looking at the Adafruit library:
> https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library
> They have an example program that [...]

I have been working for some time now, trying to make a PCA9685  driver for the same. I'm new to C++ and would appreciate it if you could explain a few parts of the code where I'm facing problems so that I can finish the driver.
TIA

Auc7us, 2026-08-29:
Claude / GPT should handle it like a breeze, im not sure if my driver from back then still works , you can try
https://github.com/Auc7us/Auc7us_v0.1/tree/master/robot