KurtE/Arduino_Phoenix_Parts #3: additional support requested

machinehistories, 2018-01-07:
I really appreciate this library and I was hoping you could provide an example file that showing ServoGroupMove.moving(); and  ServoGroupMove.wait();. Specifically I am sending servo values via serial and I was wondering how these methods could ensure that each line of serial values is properly processed before applying the next ServoGroupMove.start();. thanks

KurtE, 2018-01-10:
Sorry, I am not sure if I still have some examples of this.  I believe you are looking at the ServoEx stuff which is for RC servos, which I have not done much with for several years now. 

The file phoenix_driver_ServoEx.h as part of this project  uses it.  For example it uses it in the FindServoOffsets code to wiggle the selected joint.   At least it uses the wait. 

The only example I have of moving is the real simple Sweep test program that is an example under the ServoEx library.

I am not sure if that helps or not?

machinehistories, 2018-01-12:
Thanks the sweep sample helped but I specifically want to apply it to ServoGroups using ServoGroupMove.moving(). I will dig around more but when I do I came across this quote from you on the robotshop forum "If people are interested I should put a readme file up there. I did servoEx awhile ago as I wanted some of the same capabilities of the SSC-32 when I started adapting programs from Basic Atom Pros using an SSC-32 or the HServo code over to an Arduino. I was a simple adaptation of the Servo library. Can go into more details if interested. Also I should probably take a pass through and see if anything has been added/fixed in the Servo library that should be folded into this." Well I guess I am interested. Just kidding I know that was in 2014 so I am sorry I am sure you are working on other matters. I will at least tell you that the drawbot I made using your library and interpolation really does work so much better than the standard servo library so thanks for the code.