KurtE/Arduino_Phoenix_Parts #4: Project not open for collaboration

klimbot, 2019-12-11:
Hey KurtE, 

I'd like to be able to collaborate with you on this project.
I've been working on my own fork for a while and would like to contribute some things back.

The main reason for wanting collaborator access (specifically) is that I'd like to be able to set up github actions for this project like I have mine, and I need to be a collaborator to do that.

Thanks

klimbot, 2019-12-11:
Just realised I can't even push a branch, I get permission denied.

KurtE, 2019-12-11:
Typically with GitHub you need to fork a project, save all your changes to your fork.  Potentially in a different branch, and when you are happy with your changes you then issue a Pull Request to the project owner who then decides if they will take in the changes.

Hope that helps

closed 2019-12-12 by klimbot